emard/ulx3s #2: MANUAL.md - ULX3S passthru from f32c project link is 404

gojimmypi, 2019-02-10:
The link in the text: "To start using ESP-32 first you need to compile ULX3S passthru from f32c project" found [here in doc/MANUAL.md](https://github.com/emard/ulx3s/blob/master/doc/MANUAL.md#programming-over-wifi) gives a 404 error, pointing [here](https://github.com/f32c/f32c/tree/master/rtl/proj/lattice/ulx3s/passthru_45f).

emard, 2019-02-10:
HI

passthru and openocd links updated in MANUAL.md

gojimmypi, 2019-02-12:
@emard confirmed operational link now points to [your ulx3s-passthru](https://github.com/emard/ulx3s-passthru) instead of the the @f32c one that appears to have been deleted from [here](https://github.com/f32c/f32c/tree/master/rtl/proj/lattice/ulx3s). 

thank you

closed 2019-02-12 by gojimmypi