emard/ulx3s #5: Some questions about using an external oscillator and pins on the ULX3S

nsajko, 2019-03-08:
Greetings,

I was introduced to ulx3s (I have the ECP5  LFE5U-25F-6BG381C FPGA and ULX3S v3.0.3) as part of the FER "Digital Logic" course (where we used it with VHDL) and now I am approaching it again as a hobbyist with the aim of making a High Speed USB (over US2) microphone array. Please bear with my ignorance as this is my first electronics project.

While preparing for this project I realized that it would maybe be necessary to have a more stable and precise oscillator than the 25 MHz one included on ulx3s, so I have some related questions:

* This 250 MHz LVDS oscillator [0] looks OK to me as a clock source. Is there perhaps a reason that trying to use it with ULX3S and my ECP5 instead of the 25 MHz oscillator on the ULX3S would be a bad idea?

* Is there a reason besides cost-cutting that ULX3S does not come with a better oscillator?

* I do not know how LVDS works, but that does not matter as the FPGA takes care of LVDS after I connect it with the "test" circuit suggested in the oscillator datasheets, only presenting digital clock signals to VHDL code? At least that is how I would prefer the situation to be!

* Am I correct in thinking that to use the LVDS oscillator with the ECP5, it is needed to connect its OUT pad/pin terminal to the J1 GP12 connection (marked with "+" on the ULX3S PCB), which would be accessible through VHDL with the gp(12) signal, assuming I use your LPF? Does the same hold for OUT- on the oscillator and J1 GN12 on the ULX3S PCB and gn(12) in VHDL, and is gn(12) supposed to always be the logical compement of gp(12) in such a case?

* Do I set the IO_TYPE for the gp[12] and gn[12] to **LVDS25** in the IOBUF PORT command/statement in the LPF file?

* Can somebody perhaps explain the purpose of the "N/C" pin terminal on the linked oscillator? I think it is supposed to always be left unconnected electrically??

* Can the Ground pin terminal be connected to something on the ULX3S? Likewise, how to draw power from the ULX3S to power the oscillator and other electronics necessary for my project? (I will probably use some kind of linear voltage regulator or voltage reference to get a more stable voltage source than the switching regulators on ULX3S can provide.) I would very much appreciate a photo of the v3.0.3 PCB with the necessary pins circled.

* Related to the previous question, I think the square-looking connections (there are 6 pairs of them on the ULX3S PCB) are ground, can I connect the oscillator ground to one of them? Are they all equivalent? On another note, what are the 4 wide circular connections with big holes through them on the ULX3S PCB?

* I am not sure how exactly to set up the LVDS resistors in the FPGA, or whether I need them.

* Did anybody ever need a heatsink for the ECP5 in their projects? Maybe I will need one ...

[0] https://www.arrow.com/en/products/dsc1103ai5-250.0000/microchip-technology

emard, 2019-03-08:
HI

Too many questions at once, I'll try to summarize and provide initial "mixed"
answer and then.

If you need to connect external differential clock source, currently
only option is GP,N 12 at external pins for PCB v3.0.3.
This is the only true Differential Clock INPUT capable pin pair.
For clock output, you can use several other pin pairs though therefore
you can route output the clock to other boards if you need to synchronize them.

You can power this LVDS oscillator directly from ULX3S it has 1A available
for powering external circuits.

Usually 3.3V powered bank can also accept 2.5V differential levels as input.
ULX3S also has jumper to completely switch GP,N 0-13 pins to 2.5V level if
you really need so. In this case LEDs will also become dimmer.

you don't need to worry much about setting of internal differential
resistor terminations,
usually any settings will work, but there are also Lattice Techincal
Notes describing
setting it in detail if you really want to be 100% electrical compliant

Squared PCB pin pads are GND, they are also labeled with "T" shaped
symbol, you have
multiple choices at each header, any will work. Headers have also 3.3V
pads "+" marked
(or they can become 2.5V, selected by the jumper between audio jack
and video connector)

at least 1A  is available to draw from the board at "+" or VCC marked pads
near GND pads. Near GP,N 27 is also +5V source directly from USB connector.

Switching regulators are noisy but for LVDS oscillator it should be no problem
as it probably has its own independent internal source.  Of course you are free
to add linear regulator if you need it but I don't think you will get
clock stability
from this.

For most stable clock source I would recommend atom clock or GPS disciplined
frequency source.

emard, 2019-03-08:
VCC +3.3V and GND are all equivalent and wired together.

ECP5 is relatively low power FPGA it would heat to about 50°C
when max loaded, so heatsink is not needed, save it for Artix-7

closed 2019-07-15 by emard