emard/ulx3s #6: constraint file pin numbering comments for gp[]

gojimmypi, 2019-03-16:
Regarding the comments for the gp pins:

I'm wondering if perhaps there's a mistake in the numbering in the comments, for example, [in the constraint file](https://github.com/emard/ulx3s/blob/master/doc/constraints/ulx3s_v20.lpf#L389):

```
LOCATE COMP "gp[14]" SITE "U18"; # J2_5+  GP14 DIFF ADC1
LOCATE COMP "gn[14]" SITE "U17"; # J2_5-  GN14 DIFF ADC0
```

specifically, I would have thought that  `J2_5+` and `J2_5-` should instead be labeled `J2_14+` and `J2_14-` (the differential numbering in the middle of the connector; J2_5 appears to be the J2 connector pin number for GN14 = U17) 

or perhaps the comments should be `J2_5+` and `J2_6-` if using connector pin numbers (then what does the `14` in the middle represent?)

from the schematic:
[image]

If not, then what exactly does the `5` mean in  `J2_5+` and `J2_5-` ?

emard, 2019-03-16:
HI

this was remains of my pin numbering required for kicad
to "understand" differential so my "logic" was to use odd
connector pins like j1_5 for the pair j1_5+ j1_5-

problem is with kicad, it would not "understand" that
j1_5+ j1_6- belong to the same differential pair because
in kicad, differential pins must be named the same except
being different by trailing - or +, then kicad autodetects this
as "pair".

But anyway, just ignore this J1_x+ pinning, use GP,N which is more logical

gojimmypi, 2019-03-16:
hm. yes, I see. I think I'll add some notes for clarification. PR coming soon.

but on this topic, I'm wondering if perhaps the `- +` label on the PCB for J2 might be reversed?

Note from the pic above, that differential pair 14 is near the end of the connector closest to Pin 1. As the 14 is near the bottom of J2, I assume this means the physical board numbering is reverse of that on the schematic, right? (this would make sense, since it is rotated 180 degrees as compared to J1.

[image]

However, if indeed Pin 1 of J2 is at the bottom of the PC board near the `3.3+`,  that would make pin one, and all the odd numbers of J2, the negative side of the differential, right?

Note the tiny `- +` printed on the board between the `14` and `15` of J2. 

For clarity, here's J2 schematic upside down, but oriented the same as viewing the PC board with the USB and DVI connectors at the top:

[image]

The negative side of the differentials should be to the right, correct? (odd J2 pin numbers on the bottom of the connector as view in the pic)

emard, 2019-03-17:
Pin 1 side left/right will be switched depending if connector
soldered is MALE or FEMALE board accepts both so some user
can solder MALE and find pinout OK, some user can solder FEMALE
and found pinout mirrored, or vice versa.

silkscreen markings on PCB should be self explainatory.
Outside pin row is differential +, inside pin row is differential -

Schematics symbol doesn't exactly map to physical appearance
of actual connector pins. I should have them mirrored/rotated probably
:)

gojimmypi, 2019-03-17:
yes, I did find that the _silkscreen on the board **is** correct_ - but the schematic is not? It does appear all of the _**even**_ numbers are G**N**xx; (as indicated by the `- +` on the J2 silkscreen) and the **_odd_** pin numbers, closest to the edge of the board are G**P**xx (of `J2`, I did not check `J1`). 

For me, this became most evident when my AD/DA was not doing any conversions. I found the CLK on pin 29 of my board, needed to be connected to [GN14 on pin 6](https://github.com/gojimmypi/ulx3s-adda/blob/development/ulx3s_v20.lpf#L492), even though pin 6 (on the inverted J2 image, above) is labeled as G**P**14.

It is certainly good news that the silkscreen is correct. Unfortunately I was working from the schematic for quite some time before I realized that. 

I've certainly had my share of _where is pin 1_ issues over the years... ;)  however that was usually with the cable. Regardless of male or female connector, pin one on the PC board remains the same, no? I do believe all the P/N notations on the schematic are reversed - and it would be really great to also invert the connector on the schematic to reflect the inverted orientation on the PC board, do you agree?

As further evidence, check out this view of the KiCad, J2 from above - viewing front of board:

[image]

And here it is rotated 180 degrees:

[image]

Note this view is from the top of the board. The only way that pin 1 would be on the left when looking from above, is if in fact the connector was on the back of the board. Do you agree? (see also [this pic](https://en.wikipedia.org/wiki/Pin_header#/media/File:Stiftwanne2x13v2.jpg) on wikipedia)

Also: Perhaps on the next version of silkscreen, you could add something to make pin 1 abundantly obvious?

If you agree with all this, I'd be happy to implement and submit a PR :)

emard, 2019-03-18:
It is possible that I made a male/female mismatch of schematic
symbol for header and the footprint of the header.
Kicad allows to mix everything and supposes the user knows what to do.

My personal preference is that pin numbering is for the straight male pins.
Then pin 1 of the cable (suppose you use IDE/CDROM 40pin cable) will
match pin 1 at the board markings

WHen using 90° female angled connectors, then usually nobody has cable
with male pins that can plug in female headers. It is
supposed to plug PMOD male connectors which have completely
unexpected pin numbering you don't want to know...

emard, 2019-03-18:
I'm looking again at the schematics -> I'd say pins on both
schematics and pcb footprint are consistenly numbered for
female 90° angled connectors.

If you soldered straight male pin header, even and odd numbering
will swap sides.

gojimmypi, 2019-03-18:
Well, I don't mean to be so persistent that I am annoying - but respectfully, I disagree. :) I'm also certainly willing to be wrong.

For one, I do not believe that changing between male and female headers changes the numbering scheme on the board. I think the reversal you are thinking of, is looking at the cable connector, or the device connector.

First, do we agree the "Front" of the ULX3S is as labeled in KiCad - the side with the display and connectors?

Let's take a look at another example, the J8 header mounted on the front of the board for the Raspbettery Pi. See the [Schematic](https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/schematics/rpi_SCH_3bplus_1p0_reduced.pdf):

[image]

Pin 1 is in the same position on the physical board of the RPi, (to the left when looking down at the connector on the front of the board) regardless of male or female connector that may be soldered on, as well as regardless of straight or angled. Agreed?

And [here's](https://github.com/raspberrypi/documentation/blob/master/usage/gpio/README.md) where they document Pin 1 as the 3v3 for the header, which would be Pin 1 for a male or female connector when the header is mounted on top of the board:

[image]

For reference, [here's a Molex angled connector from Mouser](https://www.mouser.com/datasheet/2/276/0877604016_PCB_HEADERS-176440.pdf):

[image]
[image]

I drew the red connection showing pin 1 reference, assuming the same orientation as the RPi connector, above. In both the case of looking down at the PC board, or looking at the connector, Pin 1 is to the left.

I think the issue with ribbon cables, is more whether they have the connector crimped on upside-down or not.

Here's a random female connector ribbon cable from superuser (which I agree are the most common and the headers should be male). In this case, I assume we are looking into to holes of the connector, and is completely consistent with the numbering shown above.

[image]

Note how Pin 1 _is to the right_ when looking at the cable connector, but _to the left_ looking at the device header (PC Board ). You'd flip this cable upside down to plug it in to the Molex sample above, and both Pin 1's would align.

So returning to our example of the ULX3S, the only way for Pin 1 to be to the right on the PC Board here, assuming we are looking at the front of the board, is to have the header _mounted on the back_ of the board:

[image]

Note for the cable pic shown above, the only way for Pin 1 to align is to plug the cable into the bottom of the board, exactly as oriented in both pictures.

My apologies if you believe I am wrong... I'm only looking for accuracy here - and to hopefully help others others avoid the frustration I had over the weekend. If you still believe that I am incorrect - I am certainly willing to listen, if you can show me otherwise :)

gojimmypi, 2019-03-18:
Aha - you know... I stand corrected. Indeed if the on-board connector is _female_, Pin 1 is to the right as shown in this [Mouser datasheet](https://www.mouser.com/datasheet/2/1/ts0689-66588.pdf)

[image]

I was too fixated on the standard [male header](https://www.mouser.com/datasheet/2/1/TS2244-35767.pdf), above - and even missed this part when you mentioned the PMODs (which are male and expect a female header). 

[image]

I wonder if the header should be male like most dev boards, particularly with something that looks like a ribbon cable connector, with PMOD adapters? That way end users can have breakouts like the ones for the Raspberry Pi's such as [this one from Adafruit](https://www.adafruit.com/product/1989)

Well, thanks for checking. :)

gojimmypi, 2019-03-18:
But wait! (oh no, I think I am going crazy) The original issue was regarding the _reversal on the schematic_. Note on the schematic, Pin 1 is drawn to the left (the standard for male header), but on the PC Board, Pin 1 is on the right (expecting a female connector).

My whole point was to be consistent, as I was following the schematic, incorrectly assuming it was also a physical wiring diagram - but note the P/N for #14:

[image]

The original issue was how the silkscreen `- +` _is_ correct. Note the first message for this issue (above): you comment in the constraint file that `SITE "U18"; # J2_5+ `  but the positive differential is actually on Pin 6, no? Pin 5 is labeled on the schematic as `GN14`

emard, 2019-03-18:
this is my "conversion scheme" for J+-
J2_5-  -> J2_5
J2_5+ -> J2_6

The silkcreen markings +/- should be consistent with digital
differential +/- for FPGA, but are OPPOSITE for differential mode
of ADC

if you use ADC in differential mode, it is of course connected
shared on pins GP/N14-17. What is + for digital FPGA pins is -
for ADC in differential mode and vice versa

However (I now not sure but I _think_ that) binary numbers obtained
from ADC are also reversed. Connecting + voltage to ADC+ and - voltage
to ADC- will obtain lower hex values, and reverse the higher values.
So now wonder have I reversed reversed - + and got correct HEX? :)?

You know what's difference between good and bad scientist?
Good scientist makes wrong sign even number of times :)

gojimmypi, 2019-03-18:
Well, I certainly appreciate your patience here... Clearly I've not been very effective in communicating the issue. ;)

First: Ironically, despite the GP/N14 differential pins in question being _ADC pins_, I am not actually _using_ them for that. My AD/DA converter is _external_, connected to the ULX3S as shown [here](https://github.com/gojimmypi/ulx3s-adda/blob/development/doc/ribbon-cable-ADDA-bottom.PNG) and in operation [here](https://twitter.com/gojimmypi/status/1107054057125040128) - having nothing to do with the fact that those ULX3S pins are also differential A/D pins (I was referring to the general topic of differential P/N for all naming of the pins).

The issue I am trying to convey is the schematic has Pin 1 reversed, shown on the left, and the PC Board has Pin 1 on the right:

[image]

So for instance when looking at the schematic for GN14, I connected my wire here:

[image]

Which I thought was this position on the header, Pin 5:

[image]

But the PC Board (Female connector as you noted) has that numbered as Pin 6 for both J1:

[image]

and J2:

[image]

Pin 6 is actually G**P**14, correctly noted on schematic, but illustrated on the opposite side when actually viewing the connector.

emard, 2019-03-18:
Your image snapshots are referring to old schematics

New schematics have mirrored/rotated the connectors
that in my kicad and PDF now look the same,
are numbered the same and are rotated to visually
match the same appearance as physical pins when
viewed from above of the PCB (buttons side up).

I hope now this is easier to read. Tell me if you still can
find any difference of mis-leading facts?

gojimmypi, 2019-03-18:
old? _OLD_!? LOL. You mean "old" as in the [version prior](https://github.com/emard/ulx3s/commit/2dba7d3caa8b1b38d21afe40399291c0e050c252#diff-1bc2b848c551ed479ade97c29b69f1ce) to the one you [edited just yesterday](https://github.com/emard/ulx3s/commit/a639b5e2f4d1e8efecb6504374c8083047c69553#diff-1bc2b848c551ed479ade97c29b69f1ce)?? ;)

well then: yes! that's great! :) Just like that. (and yes, I feel rather silly for ranting on above, but in the end, it appears you agreed with me that the _old_ schematic was misleading).  

I did see your changes yesterday regarding default page orientation and text searching - both excellent updates - but I had not noticed the connector changes! 

This is great, much more helpful:

[image]

[image]

Thank you for making those changes. :)

emard, 2019-03-18:
Yes it's OLD there were few updates today :)
Pin numbering remains but it's now rotated and mirrored
for a more intuitive reference to the PCB.

gojimmypi, 2019-03-18:
yes, indeed. :)

Regarding this comment, from above:
> this was remains of my pin numbering required for kicad to "understand" differential so my "logic" was to use odd connector pins like j1_5 for the pair j1_5+ j1_5- problem is with kicad, it would [...]

I wonder if perhaps since this was _your "logic"_ - perhaps the notation should be removed from the constraint file?  (using the rationale that if it was confusing to me, it may be to others as well?) Or clarified: `J1_5_6+` and `J1_5_6-` - particularly now that the note had been added to the schematic about the change in numbering based on a male or female header.

I do agree `GP,N` is more logical.

btw - you may with to add some text to the schematic that states _why_ the default as shown is a female connector, specifically to accommodate  PMOD connectors without an adapter.

emard, 2019-03-19:
HI

I prefer to keep Jm_n+- notation for now, in the constraints
file before GP/N I have added comments with examples
that describe how to convert J1_5+ to J1_5 or J1_6, depeding
on connector type.

gojimmypi, 2019-03-19:
Hi, ok.

so while I have your attention... :)

can you tell me the significance of `GR_PCLK3_0`

[image]

Is this simply a work in progress? Other pins have similar labels that could be added (and imho, would be cool to have them indicated). Just wondering if there's any importance to this particular one?

emard, 2019-03-19:
This is GR_PCLK a single ended
General Routing Primary Clock capable pin
It's the only clock capable pin at J2
It can receive or transmit clock but with small delay

For minimum possible delay there are pins labeled as PCLKT/C
those are true Primary Clock capable differential pins more of them
are on J1

gojimmypi, 2019-06-06:
hello @emard - I noticed this issue from several months ago is still open. I think you [resolved this with the latest schematic updates](https://github.com/emard/ulx3s/issues/6#issuecomment-474112359)? As I opened this issue - are you waiting on me to close it or is it left open for some other reason?

cheers

emard, 2019-06-06:
Yes close it :)

gojimmypi, 2019-06-06:
thank you for the schematic updates :)

closed 2019-06-06 by gojimmypi

emard, 2019-07-25:
HI

I ordered one
AD9280 + AD9708 AD / DA Module for AX series FPGA Development Board
and I'm thinking of adapting hdl4fpga scopeio for this ADC
do you have some extra PCB adapter to send me?

gojimmypi, 2019-07-25:
Hello @emard - yes, for you - certainly! Thank you again for all your help and patience as I learn some of this stuff. Send me a gmail dot com; email same as my github name. I have extra headers that I can include and/or solder on for you, if you'd like.

For reference, my [KiCad docs for this board](https://github.com/gojimmypi/ulx3s-adda/tree/development/adapter_board) are still on the development branch of my [ulx3s-adda](https://github.com/gojimmypi/ulx3s-adda) project. Which reminds me: I think I need to make my OshPark project public. I was really quite happy with myself in getting this [initial pass-through](https://github.com/gojimmypi/ulx3s-adda/blob/master/ulx3s-adda.v) working:

[image]

I am  looking forward to seeing what _you_ can do with that AD/DA board! The [ScopeIO](https://github.com/hdl4fpga/hdl4fpga) project looks very cool.