emard/ulx3s #7: schematics pdf is not searchable

dpavlin, 2019-03-17:
It seems that schematics pdf is not searchable, which is shame, since finding components is much easier using search. Just re-creating schematics from current version of KiCad wil fix it (tested with KiCad 5.0.2+dfsg1-1 on Debian sid.

This will also create color pdf with correct orientation, so win, win :-)

emard, 2019-03-17:
J1/J2 mirrored to look the same as on PCB.
Schematics exported from "Plot" menu is now searchable.

Before I was exported it from "Print" menu and that was
the main reason of unsearchability.

gojimmypi, 2019-03-18:
this is a great change! and thanks for orienting the schematic to be readable in landscape mode, too :)

closed 2019-03-19 by dpavlin