emard/ulx3s #9: Future 10 MHz ADC

hdl4fpga, 2019-07-15:
Is it possible to add a 10 MHz ADC in future versions ?
I think it would be very useful, as DVB-T and ISDB-T both are digital broadcasting system that are  10MHz-bandwidth OFDM signal making the ULX3S a very professional low cost DSP lab.
DVB-T is the European standard while ISDB-T is the open standard that is used in Japan and South America.

emard, 2019-07-15:
Most easily it would be to add it external, connected to one side of
2.54 ports which will for example be such 10 MHz ADC from ebay
maybe 20-50$ cost

on ULX3S we are running out of free pins...
PCB must support all chips from 12F to 85F and therefore it uses
only those pins that are common to all 12,25,45 and 85F BGA-381 packages.

There is only about 10 such free pins left and they are mostly in the center
of BGA, this area is nearly inaccessible with the low cost 4-layer PCB.

hdl4fpga, 2019-07-15:
Ohhh ... I got it 

thanks

closed 2019-07-15 by hdl4fpga

gojimmypi, 2019-07-23:
@hdl4fpga I've hooked up the ULX3S to an external AD/DA board; see [blog](https://gojimmypi.blogspot.com/2018/06/ad9280-ad9708-ad-da-module-for-ax.html) and some brief [notes](https://gojimmypi.blogspot.com/2019/03/ulx3s-sdr-with-alinx-adda.html). I also have a custom OshPark board to connect them. 

See also: https://github.com/gojimmypi/ulx3s-adda  ([development branch](https://github.com/gojimmypi/ulx3s-adda/tree/development) has [PCB Adpater Board](https://github.com/gojimmypi/ulx3s-adda/tree/development/adapter_board))

All not as polished as I might like, but perhaps you'll find something useful.

hdl4fpga, 2019-07-23:
It looks nice!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gojimmypi, 2019-07-23:
@hdl4fpga  - if you go down that road and buy one of those AD/DA boards, I can probably send you one of the adapter boards, if you want.

hdl4fpga, 2019-07-23:
Thanks

I have one old NU Horizons' SPARTAN 3A DSP  board with a LT Dual 14-Bit,
125Msps 500Mhz. When I proposed to change the ADC, I was thinking on
replacing the ones that are already on the ULX3S board. I haven't had
enough time to see it in details yet

Regards

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>