emard/ulx3s #11: Git Tags for Board Versions

bluesceada, 2020-07-30:
Is it possible to get git tags for respective board versions? We have earlier versions of the board, and it is not so easy to find the correct documentation.

emard, 2020-07-30:
That's true, I should train myself to tag releases
(It will be my first-time use of git options :)

Let me know what board you have, I think it only differs in some
LED control lines of usb-serial chip, very unimportant for average use.

Online PDF schematics its still for v3.0.x boards on my git which
are currently circulating everywhere.
kicad project online is for v3.1.1 - ready for prototyping.

v3.1.1 is because of issues with ESP32, currently workarounded in
software while clean solution is rewiring pins.
ESP32 is rewired to allow use of its RMII ETH interface and to
get rid of SPI-JTAG glitches that happen to esp32ecp5.
Clean esp32ecp5 script ready for v3.1.1 is here:
https://github.com/emard/esp32ecp5/blob/master/new/ecp5.py

emard, 2020-07-31:
Releases tagged.

closed 2020-07-31 by emard

bluesceada, 2020-08-12:
I don't see the tags, did you push them?

emard, 2020-08-12:
UPS!

I tagged on my development computer and obviously forgot to push
this week I'm away that computer so I can push earliest next week.

If you have some board and need to know which commit it belongs
to I can help

bluesceada, 2020-08-12:
We have v3.0.3 boards and it would be nice to have the right schematics for it. The whole git commit history and/or file hierarchy is hard to understand for me regarding what belongs to which version.

emard, 2020-08-12:
For v3.0.3 version you can use v3.0.8 PDF schematics
https://github.com/emard/ulx3s/blob/master/doc/schematics_v308.pdf
which is online in the current git. Your board has exception that
FTDI_nRXLED is not connected to FPGA on your board and
this is noted "NC <v3.0.5" on the page 6 of PDF at upper right corner,
quadrant A1

bluesceada, 2020-08-12:
Thanks, that helps (me at least). Your tags will hopefully make it easier to find for others later.

emard, 2020-08-12:
I think I did git tag and push but as tags were
done backwards in time maybe github didnt accept
or I missed some option, I will see and push them somehow

bluesceada, 2020-08-13:
As far as I know pushing tags can be a bit complicated with git. Especially if they are not annotated tags (tags with an additional descriptive message). So you might need to push them separately. To push all tags you have, you could use `git push --tags`. It is not recommended (since it could be a lot), but I guess it will do the job.

emard, 2020-08-13:
Yes tags I made were not annotated tags
I did simple "git push" and thought it would do the job

I should have cloned the respository in other directory
and check did the tags made it there but that I didnt do
next week I get to the PC which tagged them and will
retry

emard, 2020-08-17:
Last time I did just "git push" and tags were not sent.
Now I did "git push --tags" and it seems to work,
I cloned to /tmp directory and tags are still there!

Not all releases were tagged, only those that I know
have been produced and maybe few others what
could/should have been produced.