emard/ulx3s #14: Sample code for using LAN8720

developandplay, 2021-05-31:
Hi @emard,
thank you for creating such a good ulx3s guide.
I did get the LAN8720 module to try out ethernet on the ulx3s.

First of all is there anything I need to bear in mind regarding the pyhsical connection to the ulx3s?
Right now I just plugged it in from 9 to 13 like you mentioned in the README.

Second do you have any example code for using the module?
I tried using it in Litex but I assume I wired somthing up improperly as the my code doesn't work as expected.

emard, 2021-06-01:
HI

saxonsoc binary linux from lawrie at github should use LAN8720 out of the box.
It will autodetect eth at boot it and you can start ssh and ping
around.

LAN8720 is directly pluggable to ULX3S. You can either
solder FEMALE 90° connector on top side on ULX3S and plug module
or solder MALE 0° (straight pins) on bottom side of ULX3S and use
old CDROM/IDE 40-pin flat cable (female both sides) to connect to LAN8270 module

LAN8270 is only for 3.3V power GP 9-13. If you connect it to 5V side, it
must goto trash.

emard, 2021-06-01:
Here is also elementary verilog example which will
do some elementary packeting on the module and leds
blinking but don't expect full TCPIP

https://github.com/emard/ulx3s-misc/tree/master/examples/eth/rmii/proj

gojimmypi, 2021-06-07:
@emard this is a very cool example... thanks for sharing! I can confirm [it works](https://gojimmypi.github.io/gridster-jekyll-theme/ulx3s-fpga-ethernet/)! :)

[image]

emard, 2021-06-07:
eheheeee, great testing, thanx!!
I was also happy when LAN8720 started working,
thanks to proper readme I could do it too, after almost a year :)

I had some hacks with init sequence for ST7789
to make it more reliable with some displays with
green tab stickers (probably some very early chip
releases).

emard, 2021-06-07:
Actually you have green tab too :)

developandplay, 2021-06-08:
I got to try the sample today and it works nicely.
One thing that I noticed is as you mentioned in the project readme it's crucial to first connect VCC and GND.
This can be achieved by plugging it in slightly angled.
Powering on the FPGA with the LAN8720 already plugged in doesn't work for me.

Thanks for providing the sample now I will try to get it running in Litex and report back.

emard, 2021-06-09:
HI

Modules LAN8720 from ebay are slightly different there exist at least
2 visually different (bottom side is easier to make difference).

The problem is that module needs some "normal" power up default which means
pullups and pulldowns if when plugged to ULX3S at early boot time must keep the
same states as modules hardware default, which is not always possible because
FPGA takes time to boot from SPI flash so correct pullups are enabled
but too late, module already booted in wrong mode (usually 10Mbit or wrong clock
direction)

module has its own MDIO control interface,.that's similar to SPI and I
think all registers
can be set later. Probably a small verilog ROM initializer to just
write registers to
"normal" default should be good enough but my code doesn't have it yet.

emard, 2022-04-22:
> really nice example! Is the recommendation for using GP 9-13 also related
> to the fact that the 50 MHz RMII clock must end up at a PLCK capable pin?
> Because it does, but GN12 is part of the differential PCLKC7_1/PCLKT7_1
> and I find the Lattice documentation on those quite confusing...

yes yes its pure chance that clock capable pin matches lan8270 module (I
didnt know about this
module when I was routing PCB :). Because of confusing docs there are some
other misroutings
but practicaly everything turned out to work well so the board was followed
with luck :)

ozel, 2022-04-22:
thanks a lot for your quick response! This means the PCLK pins are mostly needed for using the fancy DDR blocks, I guess.

I really enjoy how well thought out this board is, it doesn't look just like lucky coincidences to me. :-) It's probably the most sophisticated dev board I had in my hands. I'll be looking into high bandwidth transfers between ESP32 and ECP5. SDIO and RMII seem nice candidates.

emard, 2022-04-23:
Thanx you like the board! It has been designed to have
some interesting parts with onboard routing not available on market.

ESP32 is practical tool but not too fast, on my ftp and wifi usually has
100-300 KB/s (1-3 Mbit/s) transfer to SD card or ECP5 from micropython.
New boards v3.1.7 routing supports ESP32 to RMII to external LAN8270
module (I tested that) or RMII ECP5 (linux or similar) should also work
but I have yet no demo example, however speed is still 100-300 KB/s
transfer rate on ESP32 RMII (tested with micropython and SD card)

It is very practical as connectivity feature.
Theoretically it can be faster maybe there's some optimizations
compile binary instead of micropython, router settings etc.

HAZ247, 2025-06-25:
Hi I tried to make transmitter on VHDL using the same logic but it is not working can some one help!
Code
{
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity arp_reply_fixed is
    Port (
        clk_25mhz  : in  std_logic;
        btn        : in  std_logic_vector(6 downto 0);
        gp11       : in  std_logic;
        gn11       : in  std_logic;
        gp12       : in  std_logic;
        gn12       : in  std_logic;       -- PHY 50 MHz clock
        gn9        : out std_logic;       -- RMII TXD1
        gp10       : out std_logic;       -- RMII TXD0
        gn10       : out std_logic;       -- RMII TX_EN
        gp13       : out std_logic;       -- MDC (held low)
        gn13       : out std_logic      -- MDIO (tri-stated)
    );
end arp_reply_fixed;

architecture rtl of arp_reply_fixed is
    signal rmii_clk     : std_logic := '0';          -- Clock from PHY (50 MHz)
    constant REPLY_LEN    : integer := 82;
    -- Total RMII cycles = REPLY_LEN bytes × 4 (2 bits/cycle)
    constant TOTAL_CYCLES : unsigned(12 downto 0) := to_unsigned(REPLY_LEN * 4, 13);

    type packet_array_t   is array (0 to REPLY_LEN-1) of std_logic_vector(7 downto 0);
    signal arp_packet     : packet_array_t;

    signal txindx         : unsigned(12 downto 0) := (others => '0');
    signal tx_shift       : std_logic_vector(7 downto 0) := (others => '0');
    signal tx_en          : std_logic := '0';
    signal rmii_mdc       : std_logic := '0';
begin
    ----------------------------------------------------------------------------
    -- Concurrent assignments
    ----------------------------------------------------------------------------
    -- Use PHY REFCLK directly for RMII clock
    -- (you could also rename rmii_clk to gn12 and use it in processes)
    -- Hold MDC low (we’re not using MDIO management)
    rmii_clk  <= gn12;          -- 50 MHz from PHY
    gp13 <= rmii_mdc;

    ----------------------------------------------------------------------------
    -- Packet initialization (runs once at elaboration)
    ----------------------------------------------------------------------------
    arp_init : process
    begin
        arp_packet <= (
        x"00", x"00", x"00", x"00", x"00", x"00", x"00", x"00",  -- Preamble (first 8 bytes often discarded)
        x"55", x"55", x"55", x"55", x"55", x"55", x"55", x"d5",  -- Ethernet preamble + Start Frame Delimiter (SFD)

        -- Ethernet Header
        x"ff", x"ff", x"ff", x"ff", x"ff", x"ff",                -- Destination MAC: FF:FF:FF:FF:FF:FF (broadcast)
        x"00", x"40", x"00", x"01", x"02", x"03",                -- Source MAC: 00:40:00:01:02:03
        x"08", x"06",                                           -- EtherType: 0x0806 = ARP

        -- ARP Payload
        x"00", x"01",                                           -- Hardware type: Ethernet (1)
        x"08", x"00",                                           -- Protocol type: IPv4 (0x0800)
        x"06",                                                  -- Hardware size: 6 (MAC address length)
        x"04",                                                  -- Protocol size: 4 (IPv4 address length)
        x"00", x"02",                                           -- Opcode: 2 (ARP Reply)
        x"00", x"40", x"00", x"01", x"02", x"03",               -- Sender MAC address: 00:40:00:01:02:03
        x"c0", x"a8", x"12", x"80",                             -- Sender IP address: 192.168.18.128
        x"ff", x"ff", x"ff", x"ff", x"ff", x"ff",               -- Target MAC address: FF:FF:FF:FF:FF:FF
        x"c0", x"a8", x"12", x"80",                             -- Target IP address: 192.168.18.128

        -- Padding / Ethernet minimum frame size enforcement
        x"00", x"00", x"00", x"00", x"00", x"00",               -- Padding (6 bytes)
        x"00", x"00", x"00", x"00", x"00", x"00",               -- Padding (6 bytes)
        x"00", x"00", x"00", x"00", x"00", x"00", x"00",x"00",              -- Padding (6 bytes)

        -- CRC / Trailer (may or may not be used in simulation)
        x"6d", x"2a", x"fe", x"d9"                              -- Frame Check Sequence (CRC32)

        );
        wait;  -- suspend forever
    end process arp_init;

    ----------------------------------------------------------------------------
    -- Transmit process (serializes 8-bit words to RMII 2-bit output)
    ----------------------------------------------------------------------------
    tx_proc : process(rmii_clk)
    begin
        if rising_edge(rmii_clk) then
            if txindx /= to_unsigned(REPLY_LEN * 4, txindx'length) then -- fix expression
                if txindx(1 downto 0) = "00" then
                    tx_shift <= arp_packet(to_integer(txindx(12 downto 2)));
                else
                    tx_shift <= "00" & tx_shift(7 downto 2);  -- Right shift by 2 bits
                end if;
                tx_en <= '1';
                txindx <= txindx + 1;
            else
                if btn(1) = '1' then
                    txindx <= (others => '0');
                else
                    tx_en <= '0';
                end if;
            end if;
        end if;
    end process;

    ----------------------------------------------------------------------------
    -- RMII outputs
    ----------------------------------------------------------------------------
    gn10 <= tx_en;          -- TX_EN
    gp10 <= tx_shift(0);    -- TXD0
    gn9  <= tx_shift(1);    -- TXD1

    ----------------------------------------------------------------------------
    -- MDIO tri-state (unused)
    ----------------------------------------------------------------------------
    gn13 <= 'Z';

end rtl;

}