emard/ulx3s #15: SD card detect pin

developandplay, 2021-06-10:
Does the ulx3s feature an SD card detect pin?
I looked at the pinout but couldn't find  it under that name...

emard, 2021-06-10:
No - even new PCB v3.1.7 design doesn't have CD signal connected.
PCB is very crowded, hardly no free routable pins.
with some nontrivial work and redesign it should be possible to
connect CD to FPGA, I think there's 1 free pin around SD but
difficult to route

SD card driver can query card and quickly know is card inserted
without need for CD pin.
Do you have some practical situation when CD is really good for?

developandplay, 2021-06-10:
In Litex we have support for the CD pin.
This is then passed on to the Linux driver.
See this issue for reference:
https://github.com/litex-hub/linux-on-litex-vexriscv/issues/171#issuecomment-763728692

When using the ulx3s currently the same [problem](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/203#issue-816841348) occurs.

Therefore I just wanted to ask if the ulx3s has a CD pin as it would be the easiest solution.
Considering that it doesn't exist I'll think of some work around along the lines of your suggestion.

emard, 2021-06-11:
OK I understand, having CD switch would be of a practical use
with litex. There's currently no hardware switch at sockets on
v3.0.x boards

For v3.1.x board, I will check what can be done at design
but given the total market unavailability of most of chips
for ulx3s, new boards won't be there until next year anyway
so better workaround for now

closed 2021-06-15 by developandplay