emard/ulx3s #16: Example code of using SD card in SDIO mode

developandplay, 2021-06-12:
I got SPI mode to work in Litex but had no luck with SDIO mode.
Is that supported?
If yes do you have some sample code for it?

emard, 2021-06-12:
If you want to use 4-bit mode then all sd_d[3:0] lines
need to have pullups enabled from SD card side.

If you are using 1-bit (SPI) mode probably the best
is to explicitely drive 1 unused bits sd_d[2:1] = 2'b11

emard, 2021-06-13:
additional explaination

If you want to use 4-bit mode then all sd_d[3:0] lines
need to have pullups enabled from FPGA constraints.

For pullups to be active, all sd_d lines must be used in the
design (connected to some logic).

If any sd_d line is not connected anywhere the pullup
will be also removed by router/optimizer.

emard, 2021-06-13:
Similar holds for FLASH. If it is used in 1-bit mode,
unused lines must be driven 1

pullup is not enough for flash, unused lines must be driven
1. If unused lines are left floating. line crosstalk will happen
and some FLASH chips will work, but some will be confused by
crosstalk noise and will not work making FLASH boot not work
for all users and boards.

developandplay, 2021-06-15:
What does this comment mean?
> IOBUF  PORT "sd_d[2]" PULLMODE=NONE IO_TYPE=LVCMOS33 DRIVE=4; # WiFi GPIO12 pulldown bootstrapping without 3.3V efuse

emard, 2021-06-15:
SD card is shared (connected together) with FPGA and ESP32.

sd_d[2] goes to esp32 wifi_gpio12

SD default is normally all pins pull up, but factory default
ESP32 has internal pull down to wifi_gpio12 to boot properly,
that's why FPGA sets pull to none to let esp32 boot. wifi_gpio12
controls the voltage (1.8V or 3.3V) esp32 will apply to its internal flash.

ESP32 efuse can be programmed to apply 3.3V and
ignore wifi_gpio12 - that's described in the manual.
efuse is one-time-programmable, setting can't be undone
so users must be careful not to brick esp32

developandplay, 2021-06-15:
Just to make sure I understand correctly:
So my ulx3s is currently still in factory default mode.
If I just want to use the SD card with the FPGA I don't need to set the efuse right?

goran-mahovlic, 2021-06-15:
If you have board from CrowdSupply - then fuse is already set (burned) to 3.3V

developandplay, 2021-06-16:
I got mine on Mouser and the ESP32-WROOM-32 is used if that makes a difference...

emard, 2021-06-16:
You can see if 3.3V efuse is already set when esptool.py programs the esp32
it will show "VRef calibration in efuse",

Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse,
Coding Scheme None

that means that gpio12 is free to be used regardless of its powern on
pull up/down state.

Usualy problem is that some SD cards do internal pull ups and prevent
esp32 from booting or programming when SD is inserted.

to prevent user's problems intergalaktik burns efuse to
correct state before shipment