emard/ulx3s #19: Getting the board version number

szoshi, 2023-02-28:
I have purchased a new board but am unable to locate a revision number on it. It says "skriptarnica.hr" on the back where the ESP32 chip can be soldered. The documentation says that the boards sold through this website are v3.0.3. Does that apply to my board too? I cannot find any schematics for the above version. Is there any alternative version of the schematics that I can refer to?

goran-mahovlic, 2023-02-28:
Hi,

3.0.3 is quite old, could you send picture of the board?
I think @emard can find datasheet and point to lpf if you really have 3.0.3

Do you have a link from that store?

emard, 2023-02-28:
Turn board on back side "landscape"
then look approx in the middle of the board
then slightly left or right, it should be written
v3.0.8 or v3.1.7 those are most popular versions
sold. If you bought recently from mouser
should be v3.1.7

szoshi, 2023-02-28:
Hello,

mine says 3.0.3. I was unable to spot it initially. Where can I find documentation for this version?

Thanks,

goran-mahovlic, 2023-02-28:
Oh I fond one more 3.0.3 board on ebay - it is probably some student from Croatia.

Anyway I think schematics is this one as 3.0.3 and 3.0.8 d not have big difference.

https://github.com/emard/ulx3s/blob/master/doc/schematics_v308.pdf

I do not know if lpf is same as on this repository, but FER has 3.0.8 here: http://www.nxlab.fer.hr/dl/ulx3s.lpf

emard, 2023-02-28:
for most projects you can use v3.0.8 schematics for v3.0.3
https://github.com/emard/ulx3s/blob/master/doc/schematics_v308.pdf
and practically all v2.1.x and v3.0.x binaries are interchangeable

there are small differences around connecting gpio signals and
from usb-serial to fpga and esp32-gpio sharing, rarely any project
we know really needs them, if you get stuck around there as us
and we can help

you can pull git and "reset hard" to release v3.0.3 to get back
to exact state when v3.0.3 was produced