emard/ulx3s #21: New Year's Wishlist

emard, 2025-01-03:
Deer Santa 🎅 

❄ Support placing ESP32-S3 with PSRAM and FLASH.
  Allow US1 to be routed to ESP32-S3 USB pins instead of FT231X
  (When ESP32-S3 is placed, FT231X should not be placed.)

❄ Route 3 prog indicator LEDs (BLUE, GREEN, RED)
  To pins of ESP32-S3. Turning ON GREEN LED wakes up ULX3S from RTC sleep.

❄ Route 5V that powers FT231X (maybe unmounted) to 5V input of ESP32-S3. Need some
  jumper or 0-ohm resistor to remove 5V from old ESP32 which accepts
  only 3.3V

❄ Route all avialbble GPIO of ESP-32 (TX/RX serial etc) to FPGA.

❄ try to Keep some compatibility with FT231X and old ESP32-WROVER

❄ allow placing ESP32-32E instead of ESP32-WROOM, for this
  we should add top row of pins

❄ Discard support for old ESP32-WROOM. It can be still placed but will
probably not boot when top row pins are connected.

❄ BTN0 as ESP32S3 "BOOT" button
  ULX3S BTN0 - diode - GPIO0 ESP32-S3
  Holding BTN0 and powering ULX3S should allow ESP32-S3
  to enter mode that accepts firmware upload over USB
  One additional diode has to be used for this.

             1N914
    PWRBTn  --|<-- GPIO0

❄ Micro USB -> USB-C
USB-C are much better connecters, 180 deg symmetrical
Thru-hole GND shield for more mechanical strength,
micro-usb are easy to break

emard, 2025-01-03:
GPIO0 is already routed to FPGA now it can be used
to read BTN0, it is possible remove old FPGA input for BTN0 - PWRBTn
or see if we should keep it for backwards compatibility

gojimmypi, 2025-01-04:
If it is going to be a _wish_ list, perhaps you'll consider replacing the classic ESP32 with the `-C` series instead. Those are the RISC-V chips, and moving forward [Espressif is all-in on RISC-V](https://x.com/EspressifSystem/status/1517751876128673792).

emard, 2025-01-04:
You can obviously predict things and read mind remotely :)
yes from beginning we wanted esp32 with usb integrated
to skip ftdi chip, first attempt was esp32-s2 but we didn't know
how to initialize usb-jtag and available S2 modules had no
PSRAM. Now here's esp32-s3 filled with much PSRAM
and I guess with some hacking S3 can provide all we need
to say ft231x goodbye :)

Idealy would be if I find out how to configure/compile
micropython for ESP32-S3 so that its USB exposes
usb-serial with 2 additional endpoints used for JTAG.
It's HWCDC with JTAG all in S3 hardware just it need
to comehow enabled at compile time. I can compile
such setup from arduino, now Im just looking how to
compile micropython  to expose HWCDC. Currently
out of the box it exposes only usb-serial

emard, 2025-01-04:
This is what I asked at micropython github issues maybe
someone can help
https://github.com/micropython/micropython/issues/16511